balena-io/etcher #4470: balenaEtcher not working on Windows 11

jtflynn, 2025-05-25:
Tried to create bootable USB with balenaEtcher ver. 2.12. It failed continuously with "Error spawning the child process". Tried many different images and all failed. Remembered having similar issue with PEBuilder creating USB. Found that Windows 11 24H2 security does not allow these USB creators to work. Tried the same procedures above on Windows 11 23H2 and it worked every time. Hope this helps someone.

[image]

[image]

jtflynn, 2025-06-02:
Hi, it looks as though your session finished writing but the process failed in verification. My session never started writing and threw an error after about 10 seconds of doing nothing. You could try Rufus etcher which is also a free download. It works fine with Win11 24H2. Concerning your Win11 issue, if you decide to revert to 23H2 via reinstall (which, I think is the only way), you will have to go into the registry or gpedit and set Windows to not allow updating to 24H4. It will, however, continue to update version 23H2 probably until its end-of-support. The method with least amount of effort will be to use something like Rufus etcher.

Tommekeuh, 2025-06-03:
Yes I've tried Rufus which gave no error. But when I wanted to boot from the image on my microSD card it wasn't recognise ( I want to use it for the Jetson Orin Nano Developer Kit). I have ordered a new SDcard because I got the message that the files might be corrupted. I'm going to try it again with the new card with Etcher and if this gives an error aswell I know the problem isn't Etcher so I need to downgrade  like you suggest. But since this is something I haven't done before I'm not keen on downgrading. Thx for the quick reply.

jtflynn, 2025-06-03:
Another low effort solution that will alleviate switching out your system from 24H2 to 23H2, just for the purpose to run balenaEtcher, is to install 23H2 inside a virtual machine on your 24H2 system. I have tested this with VirtualBox and it worked great.  You don't even need to do any activation or configuration for the 23H2 version since you will only be using it to burn images with balenaEtcher. Unless you already have a 23H2 iso image, you can get it via a torrent since Microsoft doesn't readily make it available anymore.

flagsoft, 2025-06-16:
**It should not be that hard to include this in Balena Etcher.... but they d'ont....** 

### How to make a bootable Windows Installer USB on a Mac
https://www.youtube.com/watch?v=4-HjtRrzk64

It's basically just one file which is too BIG for FAT format. Then use a util to split it and rename this two files.
Done.