balena-io/etcher #4473: Feature: Balena Etcher LiveCD

uuf6429, 2025-05-30:
This tool is quite useful and simple to use. The only way it could be easier is if there was a live-cd version.

Here's my current use-case: I was considering installing HAOS on some MiniPC. I used Etcher to flash the haos image on a drive. BUT what I really wanted is to flash it on the internal drive of the MiniPC. Neither HAOS nor Etcher can help in this case (unless I use some other livecd, install etcher and then do the flashing).

It's somewhat surprising that no one has done such a thing already (as far as I can tell). I think this would be extremely useful.

lurch, 2025-05-31:
If you're booting from a LiveCD, and you're going to be overwriting the MiniPC's internal HDD, where exactly are you going to download the disk image to? 🤔  😉 

> It's somewhat surprising that no one has done such a thing already (as far as I can tell). I think this would be extremely useful.

P.S. It doesn't help your MiniPC situation, but the way that Raspberry Pi have solved this "chicken and egg" problem is quite clever - the EEPROM in the Pi 4 and Pi 5 have an option to download and run an embedded version of Raspberry Pi Imager, and that's then able to download and write the OS image directly (either to a microSD card, USB drive or M.2 SSD https://www.raspberrypi.com/documentation/computers/getting-started.html#install-over-the-network
See also https://github.com/procount/pinn for a similar-but-different approach.

uuf6429, 2025-05-31:
Hi!

> P.S. It doesn't help your MiniPC situation, but the way that Raspberry Pi have solved this "chicken and egg" problem is quite clever -

There's also PXE boot (seems to be similar in principle), in theory, but in practice it's a big pain in the back to set up properly.. on Windows at least.

> If you're booting from a LiveCD, and you're going to be overwriting the MiniPC's internal HDD, where exactly are you going to download the disk image to? 🤔 😉

Good question, but somehow they solved that already. In the end I made it work like so:
1. boot up Ubuntu LiveCD
2. download HAOS image with Firefox (I'm guessing it's stored in RAM?)
3. I used a tool I found preinstalled in Ubuntu... I think it was called "Disk Image Writer" or something?

Here's the funny thing: if there is an actual limit (such as RAM size or so), Balena's etch URL-to-disk feature would be extremely useful in this scenario (download X bytes, write them to disc, discard, repeat).