balena-io/etcher #4475: 2.1.2 Nothing is Happening After Selecting Flash File

Kisagi07, 2025-06-06:
[image]

- Windows 10 Pro
- Just did a clean wipe install
- Did run as administrator and without it
- Trying to make bootable usb for linux mint 22.1 cinnamon

I can still click the setting and others button like flash from URL or Clone Drive, but it just stuck here and nothing else, can't click the Flash from File again. My friend told me to try older version while waiting if there's an answer to this. Thanks.

Here's also the console if it have any help

[image]

aethernet, 2025-06-10:
Any other infos about your setup? 
I can't repro

mwq27, 2025-06-15:
Is the file you're trying to flash a .bz2 (or other zip type) file? Try unzipping it first.

dougalco, 2025-06-16:
same exact problem, win 10, ubunto .iso file

[image]

dougalco, 2025-06-17:
Weird thing was, I tried it today and it worked! Have not booted from the stick yet. Dougal

PHate99525, 2025-07-24:
> Have you guys checked Visual C++? I installed it and it worked fine now . https://github.com/abbodi1406/vcredist/releases

It worked! Thank you bro!