balena-io/etcher #4486: BalenaEtcher does not work on Debian 13

eugenialoli, 2025-06-30:
I'm running the Appimage version on the latest Debian-Testing:
./balenaEtcher-1.7.9-ia32.AppImage: error while loading shared libraries: libz.so.1: cannot open shared object file: No such file or directory

I have double checked, the library does exist on my /lib/x86_64-linux-gnu/

lurch, 2025-06-30:
> ./balenaEtcher-1.7.9-ia32.AppImage

The `ia32` there implies that you're trying to run the 32-bit AppImage...

> /lib/x86_64-linux-gnu/

...but the `x86_64` there implies that you're actually running a 64-bit Linux distro?

sgtcoder, 2025-08-09:
The .deb file doesn't work either.

iFA88, 2025-08-24:
For dependency is `libgdk-pixbuf2.0-0` marked but that is not a valid package on Deb13, the valid package name is `libgdk-pixbuf-2.0-0` what is probably already installed.

lurch, 2025-08-24:
See also https://github.com/balena-io/etcher/issues?q=libgdk

lurch, 2026-05-09:
Fixed by v2.1.5 ?