balena-io/etcher #4498: Cannot flash xz file compiled from img (.img.xz)

zxTomw, 2025-07-23:
I was trying to flash a samsung 64GB microsd card with a disk image I made from a ubuntu 22.04.5 LTS running on a raspberry pi 4. The etcher was installed on a macbook with m1 pro and 16gb ram. Using etcher, I was never able to flash the compressed image file, either made directly with the xz command or made with the [pi-shrink](https://github.com/Drewsif/PiShrink) script when creating the image. However, flashing the original img file works perfectly, and I was also able to flash xz file with rapberry pi imager.

owendswang, 2025-10-19:
Yes. I just found this problem too. It couldn't write xxx.img.xz file directly. I had to decompress it to xxx.img to write it correctly.

billmcilhargey, 2026-04-25:
This PR https://github.com/balena-io-modules/etcher-sdk/pull/334 should resolve this issue