balena-io/etcher #4507: Error: (0 , h.requestMetadata) is not a function

Bacon2942, 2025-08-11:
All the time it says: 
Error opening source

Something went wrong while opening C:\Users\user\Downloads\Iso.iso

Error: (0 , h.requestMetadata) is not a function
                                                                                                              ___
                                                                                                             |OK|
                                                                                                            

Can you guys tell me, what I did wrong?

[image]

lurch, 2025-08-11:
Duplicate of #4150

Bacon2942, 2025-08-13:
no, I have the problem too

tlindi, 2025-10-31:
Issue is still there with v2.1.4 and tar.gz ending file. Used Balena Etcher  v1.1.8 portable successfully. 

Fix this bug!

Bacon2942, 2025-11-05:
Thank you guy for helping me! :)