balena-io/etcher #4511: Ubuntu 24.04.3 LTS - Cannot install .deb file

darrenzou2000, 2025-08-21:
downloaded  balena-etcher_2.1.4_amd64.deb  (i am using AMD)

`sudo apt install ./balena-etcher_2.1.4_amd64.deb 
`

`N: Download is performed unsandboxed as root as file '/home/darren/Downloads/balena-etcher_2.1.4_amd64.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
`

? what  does sudo do then?

idkwhattoputbcimlazy, 2025-09-06:
download the portable linux version (the one located in etcher's website) since it can be executed directly with no install

alexcorvis84, 2025-10-01:
> downloaded balena-etcher_2.1.4_amd64.deb (i am using AMD)
>
> `sudo apt install ./balena-etcher_2.1.4_amd64.deb `
>
> `N: Download is performed unsandboxed as root as file [...]

That's just a simple security warning about user _apt

If you want to ignore, please perform:

```
mv ~/Downloads/balena-etcher_2.1.4_amd64.deb /tmp/
sudo apt install /tmp/balena-etcher_2.1.4_amd64.deb
```