balena-io/etcher #4518: "error starting flasher sidecar process" when trying to make a bootable usb for linux

ChrDur1, 2025-09-13:
I tried both on MacOS and Windows and i get the same error

drakebrodsky, 2025-09-25:
Try running as admin. Worked for me

emyfops, 2025-12-25:
Same on Linux 6.18.1-arch1-2, running as root gives a black screen

ksewo, 2026-02-14:
Same with v2.1.4 on linux 6.18.9-arch1-2, running as etcher with sudo produces a white screen.

ethagnawl, 2026-09-14:
@timjrobinson's solution worked for me. (UPDATE: actually the process crashed but I think it was a separate issue.) Users may want to revert those permissions after the image has been flashed using: `xhost -si:localuser:root`, though.

nscsns, 2026-09-15:
@timjrobinson +++
this actually worked for me too! thanks a lot:) // since i rarely flash anything, this one time solution works pretty well for me x)

ethagnawl, 2026-09-15:
For whatever it's worth, I never resolved the aforementioned crash. I'm on Ubuntu, so maybe that's part of the problem? Regardless, I wound up using rpi-imager (available via apt) and it worked perfectly on the first try. 

I have used Balena a number of times over the years and I greatly appreciate the project and the efforts of its contributors but using it has become increasingly frustrating and I've had to dig up some sort of workaround(s) every time I've tried to use it in the last ~6 years. :(

nscsns, 2026-09-15:
> For whatever it's worth, I never resolved the aforementioned crash. I'm on Ubuntu, so maybe that's part of the problem? 

I'm on Linux Mint, which is an ubuntu based distro sooo... it might not be an issue with ubuntu? For me it worked fine, just as balena would work on windows before switching to linux for me... 
Maybe it's just an issue with newer versions of balena..?
Sadly, i'm not nearly educated enough to tell, so, i can only guess..